jezzab/DUMLdore #3: DUMLdore not working for me

cderoche, 2017-07-22:
Trying to downgrade my Mavic Pro from 900FW to 700FW using the bin file and DUMLdore exe in Windows 10, and I'm encountering some type of JIT debug not enabled error. Here is a snap shot and the details on the error message. Was told I should raise the issue here rather than RCGroups forum... (which makes sense :) )
[image]

**Details say this:**

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.IndexOutOfRangeException: Index was outside the bounds of the array.
at WindowsFormsApplication1.Form1.btnLoadFW_Click(Obj ect sender, EventArgs e)
at System.Windows.Forms.Control.OnClick(EventArgs e)
at System.Windows.Forms.Button.OnClick(EventArgs e)
at System.Windows.Forms.Button.OnMouseUp(MouseEventAr gs mevent)
at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
at System.Windows.Forms.Control.WndProc(Message& m)
at System.Windows.Forms.ButtonBase.WndProc(Message& m)
at System.Windows.Forms.Button.WndProc(Message& m)
at System.Windows.Forms.Control.ControlNativeWindow.O nMessage(Message& m)
at System.Windows.Forms.Control.ControlNativeWindow.W ndProc(Message& m)
at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2101.1 built by: NET47REL1LAST
CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
WindowsFormsApplication1
Assembly Version: 1.0.0.0
Win32 Version: 1.0.0.0
CodeBase: file:///C:/Users/Owner/Desktop/Mavic%20Pro%20video/DUMLdore-master/DUMLdore-master/DUMLdore.exe
----------------------------------------
System.Windows.Forms
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2094.0 built by: NET47REL1LAST
CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2093.0 built by: NET47REL1LAST
CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2046.0 built by: NET47REL1
CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2046.0 built by: NET47REL1
CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2099.0 built by: NET47REL1LAST
CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
Assembly Version: 4.0.0.0
Win32 Version: 4.7.2046.0 built by: NET47REL1
CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
<system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

kilrah, 2017-07-22:
You likely have a bad dji_system.bin file.

And always update the app before reporting an issue - a check for that was added in version 1.76 earlier today.

cderoche, 2017-07-22:
Yes the DUMLdore exe is only 25KB....

cderoche, 2017-07-22:
How do I update the app? Or just look for a new version?

cderoche, 2017-07-22:
OK the page here https://github.com/jezzab/DUMLdore says v1.76 at the top, but when you download the zip file of it all the file properties on the exe still says 1.0 and it is still 25KB. Am I doing something wrong?

cderoche, 2017-07-22:
Are my bin files correct in this case? They're tiny, only 1KB.

[image]

kilrah, 2017-07-22:
Obviously not. Again you need Git LFS, out of scope of this repo/issue tracker. 

[image]

For DUMLdore jsut launch it and look at the title bar...

cderoche, 2017-07-22:
I installed Git LFS - I downloaded the bin files I wanted, now they're 120MB for Mavic roughly and 33MB for the RC. The DUMLdore exe on the Github is only 25kb.

cderoche, 2017-07-22:
I do have version 1.76 of DUMLdore, just checked.... is it only 25KB also like v1.70 that I had in the initial post? I'll give it another shot with the Mavic in a few.

kilrah, 2017-07-22:
The DUMLdore exe IS 25 Kb. Perfectly normal. No LFS in the DUMLdore repo, only ont he FW repos.

cderoche, 2017-07-22:
Excellent, hey thanks for the help, if 1.76 works I'll let you know.

jezzab, 2017-07-23:
The fault was because the firmware files were incorrectly downloaded and too small. The new 1.76 version will detect file that are too small and show an error not a hard fault like the original issue.

closed 2017-07-23 by jezzab