jezzab/DUMLdore #4: Failure on Backup- Win 10- Index was outside the bounds of the array error

JD4x4, 2017-07-26:
1st time using DUMLdore, attempting Backup of .300 Spark on Win 10 OS. Failed 3 times with same "Index was outside the bounds of the array error" error. 

Operation stops immediately after decrypting and displaying "Fixing headers", leaves form on screen if Continue is selected but is hung. Form clears if Quit is selected.

Creates backup & decrypt folders, populates with normal size files in backup, zero byte files in decrypt.
Does not create a bin file. Is it only supposed to create sig files??

1st & 2nd failures using local Git clone source files copied to 2 different folders/hds. 3rd try using dl'ed zip file from GitHub jezzab/DUMLdore master.

[image]

`See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at WindowsFormsApplication1.Form1.btnBackupFW_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2101.1 built by: NET47REL1LAST
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
WindowsFormsApplication1
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///D:/DUMLdore-master/DUMLdore.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2094.0 built by: NET47REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2093.0 built by: NET47REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2046.0 built by: NET47REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2046.0 built by: NET47REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2099.0 built by: NET47REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2046.0 built by: NET47REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2046.0 built by: NET47REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
WinSCPnet
    Assembly Version: 1.3.7.7601
    Win32 Version: 1.3.7.7601
    CodeBase: file:///D:/DUMLdore-master/WinSCPnet.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.`

jezzab, 2017-07-26:
There seems to be an issue with OpenSSL and it being called (openssl.exe). I am struggling to replicate the issue.

JD4x4, 2017-07-26:
ty. C'mon over to my house & use my PC if ya like. I've got some beers, but going out of town on Friday night, lol.

JD4x4, 2017-07-26:
Just for giggles, I ran Win compatibility checker on openssl.exe and it suggested running it in Win 8 compatibility mode, so I set it to that & tried again. No change, same error.

You sure it's not some obscure internal variable for Windoze Forms that rarely gets used & isn't getting set? (I know nothing.. just throwing stuff out there).

jezzab, 2017-07-26:
No the program downloads the sig files into /backup/ then runs openssl to decrypt them into /decrypt/ then reads them in and fixes each header. Its failing because openssl failed and the files in decrypt are 0 bytes. 
I have not been able to replicate it and no one has had this error

JD4x4, 2017-07-26:
Well. That's just the story of my life. I checked openssl.exe in both my dl'ed zip and in my local clone, both the same 536 kb and the same CRC/SHA. 

I do have a Linux Mint box so I can try a backup with pyDUML. Is openssl used only when doing a backup, or is it needed for f/w loading also? I'd hate to be totally shackled to my Linux box, but if that's my only option ..

jezzab, 2017-07-26:
Its only needed for making a backup.

JD4x4, 2017-07-26:
I'm not seeing a way to easily make a f/w backup with pyDUML. Is there a way to finish the backup steps manually using DUMLdore & tools in folder? i.e. using a cmd window in the DUMLdore folder on the sig files it's already pulled from the AC?

JD4x4, 2017-07-27:
Problem found, backup succeeded.
I opened a dos window in my working dir-
>D:\DUMLdore>openssl ?
This version of D:\DUMLdore\openssl.exe is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher.
>
>D:\DUMLdore>

Then found [https://wiki.openssl.org/index.php/Binaries](url)
where I found -

> Pre-compiled Win32/64 libraries without external dependencies to the Microsoft Visual Studio Runtime DLLs, except for the system provided msvcrt.dll.

with the link to [https://indy.fulgan.com/SSL/](url) which is a repo for compiled binaries for various OS's

Replaced openssl.exe, libeay32.dll, and ssleay32.dll with the files from there for my Win10 x86 system. You might want to distribute these instead, as they might be more universal.

Lastly, it would be great if there was an option (or switch?) to keep the decrypted sig folder & files in case anyone wants to experiment with them. This assuming that Assistant param changes might be reflected in them?

closed 2017-07-27 by JD4x4