jezzab/DUMLdore #5: Enhancement: Pull upgrade00.log

kilrah, 2017-08-01:
Since the whole mechanism to decrypt FTP files is in place, it could be useful for troubleshooting to add a button with a one-click way for people to pull their upgrade log

jezzab, 2017-08-01:
https://github.com/MAVProxyUser/DJI_ftpd_aes_unscramble

kilrah, 2017-08-01:
Sure, but DUMLdore was made for people who couldn't easily fiddle with a python script and FTP client, so it only makes sense that once they've up/downgraded using DUMLdore and still have it open they can just click a button to get the log if there was an issue...

This was in response to a thread that is now posted on DUMLdore usage on a user forum where someone had an issue - they could click the DUMLdore buttons, but them not being savvy with the other tools to get the log it makes troubleshooting difficult.

MAVProxyUser, 2017-08-01:
@kilrah  this outta do it... 
https://github.com/MAVProxyUser/DUMLrub/releases/download/v.01/LogJammer.zip
https://github.com/MAVProxyUser/DUMLrub/blob/master/LogJammer.rb

closed 2017-08-29 by jezzab