jezzab/DUMLdore #9: Timeout while trying to backup Mavic Pro v01.03.1000 firmware

musbaalbaki, 2017-08-29:
This timeout is happening every single time I try to backup my current Mavic Pro firmware v01.03.1000 firmware. Note that I tried to do it many time with full charged battery and different cables. The Mavic is connected, I can check that if I close the DUMLdore tool and open DJI Assistant 2, I can confirm the connection is working. Shall I continue to upload and flash the firmware anyway at this point? Cause I am suspecting something is wrong with my current firmware and I would like to install a fresh copy of it. I tried to install a fresh copy using DJI Assistant 2 but my issues with the drone persisted. Please help.
[image]

jezzab, 2017-08-29:
Did you try doing what the dialog box says?

musbaalbaki, 2017-08-29:
@jezzab Yep, I tried that but the browser hanged meaning it opened a blank page. What could cause FTP server connectivity to drop even though the Mavic is still connected. Also do you advise me to try flashing the firmware to install a fresh copy at this point? I mean even if the backend process is failing. Much appreciated

musbaalbaki, 2017-08-29:
@jezzab I was able to reflash successfully using DJI Assistant 2 but I don't think the reflash of DJI Assistant 2  actually install a complete fresh copy so that's why I wanted to try DUMLdore but I was afraid to try flashing using DUMLdore since the backup process is timing out. Please let me know what is best to do now. Thank you in advance.

jezzab, 2017-08-29:
You should be fine flashing. It will either upload and flash or just fail trying to upload. It wont hurt the aircraft. If the aircraft sees the file isnt there, corrupt or wrong it wont do anything.

musbaalbaki, 2017-08-29:
Great, thank you @jezzab I will try that and see how it goes. Much appreciated!

leonidas784, 2017-08-29:
Downgrade on Mavic Pro from v01.03.1000 work great, but backup was created uncorrectly. File dji_system.bin is 20KB long.

jezzab, 2017-08-29:
A copy of the incorrect dji_system.bin would be good to see

leonidas784, 2017-08-29:
1.88
Here is backup of RC 
[dji_system.zip](https://github.com/jezzab/DUMLdore/files/1260658/dji_system.zip)

kilrah, 2017-08-29:
So you were trying to backup 1000, right? They might have changed something in that version that prevents backup from working.

jezzab, 2017-08-30:
Tested backup of RC on .1000 - Test OK (slow but ok)

leonidas784, 2017-08-30:
Ok, thank you. Anyway it doesn't matter. It is more iportant that flashing always working.

closed 2017-08-30 by jezzab