jezzab/DUMLdore #10: Backing up .1000 firmware

joevoisin, 2017-09-05:
It looks like there is a new timeout being seen in FTP for some people (myself included) where I cannot backup firmware.

Recommend using a timeout value of 60 or 120 seconds when passing commands to winscp.  

Tested this with a manual copy of the file with a 30 second timeout.. it fails.. increased the timeout and it was successful.  

Dumldore tries the same firmware over and over and over.. each time it waits about 30 seconds and then gives up.. sends the FIN.. then 10 seconds later the mavic tries to start the transfer but the session was already closed... 

`24.425866	192.168.42.2	192.168.42.3	FTP	80	Response: 200 Operation successful
24.426123	192.168.42.3	192.168.42.2	FTP	60	Request: PASV
24.426855	192.168.42.2	192.168.42.3	FTP	89	Response: 227 PASV ok (192,168,42,2,177,44)
24.427676	192.168.42.3	192.168.42.2	FTP	109	Request: RETR wm220_0100_v02.08.05.04_20170802_ca02.pro.fw.sig
24.467390	192.168.42.2	192.168.42.3	TCP	54	21 ? 49859 [ACK] Seq=660 Ack=455 Win=14656 Len=0
54.005137	192.168.42.3	192.168.42.2	TCP	54	49859 ? 21 [FIN, ACK] Seq=455 Ack=660 Win=524800 Len=0
54.045314	192.168.42.2	192.168.42.3	TCP	54	21 ? 49859 [ACK] Seq=660 Ack=456 Win=14656 Len=0
63.674794	192.168.42.2	192.168.42.3	FTP	155	Response: 150 Opening BINARY connection for wm220_0100_v02.08.05.04_20170802_ca02.pro.fw.sig (37757376 bytes)
63.674885	192.168.42.3	192.168.42.2	TCP	54	49859 ? 21 [RST, ACK] Seq=456 Ack=761 Win=0 Len=0`

jezzab, 2017-09-05:
Can you please downloaded the latest version I have just pushed (v1.90) and see if it resolves the issue. The timeout on FTP has been extended from 30 secs to 120 sec. 
Its because the AC/RC on the later firmwares seems to take soooo much longer to internally encrypt the file before it is sent over FTP.
This also makes the backup process a lot slower :(

joevoisin, 2017-09-05:
I figured it was something like that.  I could see it trying about 10 seconds after the client (winscp) gave up.. 

I just tested it now and it's working fine.

Good stuff responding so quick.. Saw people in the forums asking the same question since .1000 was released.  and noticed the one video showing someone going from .1000 to .0700, they (digdat0) didn't  do the backup first.. 

the backup process was not too bad.  it looks to be consistently under 60 seconds till start of xfer.  setting it to 120 is safe then.  no reason to set the timeout low.. if it's not responding no reason to not wait..  lol..

closed 2017-09-05 by joevoisin

kilrah, 2017-09-05:
Seems it's variable from a machine to another too, the 30sec timeout was already an extension from the original 15sec and for the testers at the time it was enough...
Wonder what they changed to reduce the performance of their process so much...

joevoisin, 2017-09-05:
Not sure.  I got mine with .900 already on it and since it was already kinda nerfed, I went ahead and put .1000 on it..  just recently did I start looking at going back down to .700..   glad it was a simple fix..  they probably have some crazy process running on it soaking up all the CPU cycles.. ;)