jezzab/DUMLdore #11: Backup aircraft firmware file always invalid...

localhorst24, 2017-09-06:
...when i try to load the saved backup firmware file.
(mavic pro firmware .1000)

jezzab, 2017-09-06:
You need to provide a bit more information:
What version of DUMLdore are you running?
What is the file size of the backup?

localhorst24, 2017-09-06:
DUMLdore V1.90 (V1.90.0.0)
The file size is 110.141.440 Bytes for the [Aircraft](https://www.dropbox.com/s/p3m3csv05e64xkb/Aircraft%20dji_system.bin?dl=1) and 217.190.400 Bytes for the [Remote Control](https://www.dropbox.com/s/haadhfkwti3xftx/Remote%20Controler%20dji_system.bin?dl=1).
No Problems at all (Win10x64).

jezzab, 2017-09-06:
For some reason the files have not been decrypted correctly. 
Also it looks like the first backup file was not moved before the second backup was done.
Did you download the entire zip file?

jezzab, 2017-09-07:
Please download and test the latest version (you will need to make a new backup with it). Tested and working here. They changed the encryption keys :)

closed 2017-09-07 by localhorst24