jezzab/DUMLdore #16: Error in DUMLdore V1.92

39Den39, 2017-10-03:
Hello! I have problem - DUMLdore see Mavic Pro, when i try flash i see error - Timed out attempting to contact the FTP server on the device/ You can test for connectivity by going to  ftp://192.168.42.2/ with the browser.
 This site -  ftp://192.168.42.2/ not open in Google Chrome on PC and Android Phone. This is problem with server??  Thank you!

jezzab, 2017-10-03:
There is a problem with the connectivity of your drone or the drivers to the PC. If you can't access that link on the PC with the drone attached then you cannot flash or do anything. I would suggest trying to re-install Assistant2.

39Den39, 2017-10-03:
Jeremy, now my Mavic detected with firmware version 00.00.0000 ( after unsuccessful upgrade - flashing freeze and i power off Mavic) I install on PC latest DJI Assistant and try upgrade using official method on .1000 firmware. BUT! All freezes at 0% with "Transmitting..." This means that this Maviс can no longer be programmed at all??? I need buy new Mavic ? (((  There are still any ways to flash any firmware in to my Mavic ??

jezzab, 2017-10-03:
If Assistant2 cannot flash the aircraft then DUMLdore cannot flash the aircraft as they work exactly the same.

39Den39, 2017-10-03:
Ok. There are other ways to revive - to write firmware in drone?

jezzab, 2017-10-03:
Not that I am aware of. Why did you turn off the Mavic while it was upgrading? It can look like its paused during the upgrade as it can take up to 15 minutes.

MAVProxyUser, 2017-10-03:
Yeah that was a bone headed move, you need to buy a new core board. Or contact DJI care about your brick.

39Den39, 2017-10-03:
Progress freeze for 1 hour and battery in Mavic indicate low charge (

jezzab, 2017-10-03:
What firmware did you use and where did it come from? What version of firmware where you going from and to?

39Den39, 2017-10-03:
I have 900 version, but with night shots i often have bug with RAW photos - they damaged. Day photos is ok. I tried made downgrade to 700 version, but process is freeze....

closed 2017-10-05 by jezzab