jezzab/DUMLdore #21: backup firmware can't work properly

davidtwchn, 2017-11-29:
I have downloaded the newest version DUMLdore 2.00,backuped the firmware after the RC connected.DUMLdore shows the information "backup complete”,but the size of  dji_system.bin is only 10KB.
RC of my Mavic is 1.04.0000.
The attached file is the dji_system.bin.
[dji_system.zip](https://github.com/jezzab/DUMLdore/files/1513439/dji_system.zip)

jezzab, 2017-11-29:
-Make sure you download the entire release:
https://github.com/jezzab/DUMLdore/releases/download/v2.00/DUMLdore.zip
-Run DUMLdore.exe as Administrator
-Make sure you are running x64 Windows

davidtwchn, 2017-11-29:
I'm sorry,my Windows is 32-bit version.Backup firmware can't work properly,but Flash firmware is OK.So backup firmware can't be supported in 32-bit Windows?

jezzab, 2017-11-29:
You don't need to backup as files are already available and flashing will work fine. If you feel you must backup:
-Download https://indy.fulgan.com/SSL/openssl-0.9.8s-i386-win32.zip
-Copy the following 3 files from the openssl-0.9.8s-i386-win32.zip folder into the main folder (overwriting existing) and retry backup. openssl.exe, ssleay32.dll, and libeay32.dll

davidtwchn, 2017-11-29:
As what you said,it's not necessary to backup now,since those files are already available.  
I have unpacked these three files to the main folder,and backup firmware is OK.Thanks a lot!

closed 2017-11-29 by davidtwchn