jezzab/DUMLdore #22: .net framework exception error

ym58, 2018-01-01:
Hello,

I am having an exception error at the end of the backup process.

Apparently, the **\backup** folder (with lots of .sig inside) and the **dji_system.bin** file were both created right, but I am still a bit suspicious if the **WHOLE** backup process went well or not !

Could you tell me if everything is alright from the following snapshots ?

Thanks.

[image]
[image]
[image]

ym58, 2018-01-02:
# here's the exception error message I got  : 

_System.IO.IOException: Le répertoire n’est pas vide.

   à System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   à System.IO.Directory.DeleteHelper(String fullPath, String userPath, Boolean recursive, Boolean throwOnTopLevelDirectoryNotFound)
   à System.IO.Directory.Delete(String fullPath, String userPath, Boolean recursive, Boolean checkHost)
   à System.IO.Directory.Delete(String path, Boolean recursive)
   à DUMLdore.Form1.btnBackupFW_Click(Object sender, EventArgs e)
   à System.Windows.Forms.Control.OnClick(EventArgs e)
   à System.Windows.Forms.Button.OnClick(EventArgs e)
   à System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   à System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   à System.Windows.Forms.Control.WndProc(Message& m)
   à System.Windows.Forms.ButtonBase.WndProc(Message& m)
   à System.Windows.Forms.Button.WndProc(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   à System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

--------------------- Assemblys chargés ---------------
mscorlib
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2117.0 built by: NET47REL1LAST
    CodeBase : file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
DUMLdore
    Version de l'assembly : 1.93.0.0
    Version Win32 : 1.93.0.0
    CodeBase : file:///C:/Users/YANN/Dropbox/Public/Drone/FW%20et%20App%20misc.versions/MP/DUMLdore-master/DUMLdore.exe
----------------------------------------
System.Windows.Forms
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2117.0 built by: NET47REL1LAST
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2117.0 built by: NET47REL1LAST
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2117.0 built by: NET47REL1LAST
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2053.0 built by: NET47REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2117.0 built by: NET47REL1LAST
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2117.0 built by: NET47REL1LAST
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
WinSCPnet
    Version de l'assembly : 1.3.7.7601
    Version Win32 : 1.3.7.7601
    CodeBase : file:///C:/Users/YANN/Dropbox/Public/Drone/FW%20et%20App%20misc.versions/MP/DUMLdore-master/WinSCPnet.DLL
----------------------------------------
mscorlib.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2053.0 built by: NET47REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
System.Windows.Forms.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.7.2053.0 built by: NET47REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------_

kilrah, 2018-01-02:
Don't run this inside a dropbox dir... dropbox is likely trying to upload all the temp files so when dumldore tries to delete them it can't.

jezzab, 2018-01-02:
And please download the latest version. It's showing v1.93 which is very old. If you didnt download it from github there is no support

ym58, 2018-01-02:
@kilrah : good point, thanks !
@jezzab : it's showing vers 2.00 when I run it and I definitely DL it from your github.
[image]

jezzab, 2018-01-02:
I'm going to take a stab in the dark here and your running a 32bit version of Windows. If this is the case (based on the OpenSSL decryption failing from your pics) it wont work. you will need to download the 32bit version of OpenSSL.

ym58, 2018-01-02:
I am definitely running x64 and I DL everything from the github.

[image]

jezzab, 2018-01-02:
Ill go with @kilrah then. Copy the DUMLdore-master dir to the desktop and run it from there. Not in Dropbox

ym58, 2018-01-02:
@both of you : out of Dropbox, it now works flawlessly ... thanks alot for your support !

jezzab, 2018-01-02:
No worries. You owe @kilrah a beer lol

closed 2018-01-02 by jezzab

ym58, 2018-01-02:
Yes, I owe him also from RCgroups ... we'll have two beers, then !