jezzab/DUMLdore #24: The mavic still can't be downgrade

leoking675951264, 2018-01-03:
I have try many times from the v4.0100 .   and my friend have try too . The fm can be upload to the mavic ,but the downgrade process is ended at 80% which i see in the DJI Assistant 2

haloweenhamster, 2018-01-03:
Its a known problem with the mavic that a flash can freeze at around 80-90% (make sure you have an external fan on mavic as it can be damaged by over heating in this state) 
If you are having repeated issues ie fails to flash more than once, try flashing in steps ie 1.4.0100 to 1.4.0000 or 1.3.1000 to 1.3.700

leoking675951264, 2018-01-03:
i have successed been to the 1.3.0400  but  when i upgrade to 1.4.0100 which i changed sth , it seemed failed the red led is on  and  the middle doubled green led of battlery  is always on

zh169, 2018-01-03:
I can load 1.4.0000 then it sais it will be flash it on mavic and I can open Assistant 2 to watch the Progress but nothing happens ??

wima75, 2018-01-03:
@zh169 Please read the release notes for V2.11
https://github.com/jezzab/DUMLdore/releases
I had the same problem.

jezzab, 2018-01-03:
V2.12 was just released. Please try that and read the release notes with it. It fixes a few issues

haloweenhamster, 2018-01-03:
@leoking it sounds like the battery has failed the update, try updating in assistant 2

DUMLdore now on version 2.12 a glitch from yesterday has been sorted

leoking675951264, 2018-01-03:
@haloweenhamster  any way i finished , thank for guilding , a bug is  sometime it has to be flash twice

closed 2018-01-03 by jezzab

zh169, 2018-01-03:
I havea  Mavic Pro Platinum, is this working ??