jezzab/DUMLdore #26: DUMLdore V2.12 freezes during stage2.bin

Patapal0, 2018-01-05:
Hello Jeremy,
first of all thanks for all materials & support. If this is not the right place for this post, please let me know.

I am using DUMLdore V2.12 in order to move from firmware v1.04.0100 to V01.03.0700

I followed digdat0 video tutorial and everything started well, but now I am blocked after clicking on "Flash Firmware"
DUMLdore window is open and showing in lower left corner: "stage2.bin"
It´s been more than 10 minutes like that...
Not sure what can be done
I have already tried 2-3 times and I´m blocked on the same place...

Many thanks in any case

jezzab, 2018-01-05:
Hi,
I would suggest doing a Refresh with Assistant2 for 1.4.0100 and then attempting again

Patapal0, 2018-01-05:
Thanks a lot for quick answer!
That´s exactly what I did and everything is good now. I just finished replacing DJI Assistant 2 1.2.0 by 1.1.2 and parameters update also done! (which actually was my target)

Your job is quite amazing, thanks a lot,
Greetings from south of Spain!
[cid:image001.jpg@01D38688.CB02F800]

Enviado el: sábado, 06 de enero de 2018 0:43
Para: jezzab/DUMLdore <DUMLdore@noreply.github.com>
CC: Patapal0 <patapal0@hotmail.com>; Author <author@noreply.github.com>

closed 2018-01-05 by jezzab