jezzab/DUMLdore #28: Flashing DJI spark controller

yustme, 2018-01-28:
Hi,
I'm not able to flash spark controller. When I download the firmware for controller it's in tar file. App is not able to handle tar files. I tried to untar the files, but there are only .sig files. When app accepts only *dji_system.bin files.

How to repeat:
 - download FW fot controller with DankDroneDownloader (V01.00.0300_RC_Spark_fw.tar)
 - open DumlDore.exe
 -- select Spark Remote Controller
 -- Click on "Load firmware"
 -- it does not accepts *.tar files/.sig files

Thank you very much for your help. I really appreciate app you created.

jezzab, 2018-01-28:
In the file selection dialogue box, type
*. *
And hit enter. 
I will add the tar mask in the next release

jezzab, 2018-01-28:
Ans select the tar file. Do not untar it and flash a sig file

yustme, 2018-01-28:
Thanks very much for quick response! 
So easy ... :)

I got and error about not existing COM1. So I connected PC <> Controller with WiFi and AC <> PC with usb. And it works fine for me.

I want to donate your project but I did not find a link. Please send me the link if you have it. 

Thank you ! :)

closed 2018-01-28 by yustme