jezzab/DUMLdore #30: Stuck on "rollback defeat stage complete"

garvani, 2018-06-16:
Im following http://dji.retroroms.info/howto/nfzbypass
I have done the first part and updated the firmware.
I am now Trying step 5, after replacing the stage2 file with the one linked, selecting with alt and flashing, it transfers then i get "rollback defeat stage complete" then doesn't proceed past that. I dont see "Completed. Exit DUMLdore and restart device" Ive tried 5+ times, with DUMLdore 3.03 and 3.07. Left it for 20 mins, doesnt move past that. Tried closing and reopening etc. Dont know why its not working?

jezzab, 2018-06-16:
That patch was provided externally "as is" with no support. It's not part of DUMLdore. I would suggest NLD if you are looking for a more reliable result with your Mavic Pro.

closed 2018-06-16 by jezzab

jezzab, 2018-06-16:
No, the patch was made by me personally but it's not part of DUMLdore. I handed it out for FREE with the caveat of it is as is and people have to write a guide how to use it. I did a quick edit to DUMLdore so it could flash it to help people out. 
Hence it has nothing to do with DUMLdore.