jezzab/DUMLdore #31: Crash report

kika, 2018-06-19:
When I start dumldore with Mavic connected it doesn't run. It appears in the list of processes for a minute or so, doesn't show the window, consumes about 2.5-3% CPU and slowly gets from 20MB memory to about 45MB and then quits. When I start with Mavic disconnected it shows the window with only "About" button enabled. If I start dumldore and only then start Mavic, then it crashes:
```
************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at DUMLdoreV3.Form1.Disconnect()
   at DUMLdoreV3.Form1.USBPort_USBDeviceRemoved(Object sender, USBDeviceEventArgs e)
   at USBClassLibrary.USBClass.ProcessWindowsMessage(Int32 Msg, IntPtr WParam, IntPtr LParam, Boolean& handled)
   at DUMLdoreV3.Form1.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3110.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
DUMLdoreV3
    Assembly Version: 3.0.6740.22920
    Win32 Version: 3.0.0.0
    CodeBase: file:///C:/Mavic/DUMLdoreV3.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0 built by: NET472REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0 built by: NET472REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0 built by: NET472REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.ValueTuple
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.ValueTuple/v4.0_4.0.0.0__cc7b13ffcd2ddd51/System.ValueTuple.dll
----------------------------------------
USBClassLibrary
    Assembly Version: 1.0.5.0
    Win32 Version: 3.0.0.0
    CodeBase: file:///C:/Mavic/DUMLdoreV3.exe
----------------------------------------
DUML
    Assembly Version: 1.0.0.0
    Win32 Version: 3.0.0.0
    CodeBase: file:///C:/Mavic/DUMLdoreV3.exe
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0 built by: NET472REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3110.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0 built by: NET472REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

```

kika, 2018-06-19:
`adb shell` sees the Mavic and it's rooted, so this is not a serial communication issue

jezzab, 2018-06-19:
Yes but ADB uses a different driver to communicate. Not a virtual comport. 
What does it show in Device Manager under Ports? Should be 'DJI Device xxxxxxxxxx'

kika, 2018-06-19:
1. Windows 10 Pro N 64
2. Latest 3.07
3. For some reason it shows 2 virtual COMs - COM4 and COM5 both DJI Virtual. I tried to reboot the PC, disconnected and connected Mavic, they both disappear and then both appear together.
4. Mavic Pro (original, old one)
5. `1.04.300`
6. Assistant reports `1.04.300`

I was on `03.600`, upgraded to `04.300`, rooted with `DUMLracer`, verified with `adb` that root is OK, and then this.

Connected Mavic to my Mac and tried to use `DUMLflasher`. It also doesn't work, but in a more mysterious way - everything happens as intended, `DUMLflasher` says "Exiting....". But `dji_system.bin` sits in `/upgrade/` directory forever and nothing happens (verified with `ftp 192.168.42.2`)

jezzab, 2018-06-19:
Sounds like something strange with the DJI virtual comport driver as there should only be one. 
If it has two it will certainly confuse things. 
I would try and remove the driver and install the drivers again. Installing the latest Assistant2 (run as Administrator) should reinstall them.
The reason DUMLracer is doing that is because it has FTP access but it cant send the "start upgrade" etc commands over the serial port. 
This is a serial port driver issue

kika, 2018-06-19:
I thought so too (once I saw 2 drivers, that is). Is it possible to add port selection to the program? I just succeeded  (went to `1.03.700`) with `pyduml`, it allows to chose the port explicitly.

jezzab, 2018-06-19:
No. It's works fine when your drivers are not screwed up :)

closed 2018-06-19 by jezzab

kika, 2018-06-19:
Reinstall of the drivers didn't help. It just what it is. What seems to work (at least `DUMLdore` starts OK) is to just disable the COM5 in device manager. 
What's strange is that they indeed have different device instance paths:
COM4 (working): `USB\VID_2CA3&PID_001F&MI_04\6&30171864&1&0004`
COM5 (not working): `USB\VID_2CA3&PID_001F&MI_02\6&30171864&1&0002`

sequell, 2022-08-13:
Hello there,

I have mavic Pro, I was installing via Dumlracer to downgrade the software version. I reached up to 99% on dumldore. At 99%, the drone started to make USB plug / unplug sounds by itself. I was never able to connect again. Is the software crashed? Is there a comeback to this?