jezzab/DUMLdore #42: Issue Flashing P4P

TACK86, 2019-01-03:
I keep getting a timeout error when tryin to roll back FW on my P4P....Any insight would be great, Thank you

jezzab, 2019-01-03:
What version of DUMLdore are you running?
What version of Windows are you running?
Is it a physical Windows machine or a Virtual Machine?
What is the exact error?

TACK86, 2019-01-04:
>           What version of DUMLdore are you running?
> What version of Windows are you running?
> Is it a physical Windows machine or a Virtual Machine?
> What firmware version is the P4P on?
> What is the exact error message/dialog box?

-I've tried DD 3.1, 2.12, and 2.14
-windows 10
-physical windows machine
-firmware is 01.04.0602, trying to roll back to 01.03.0509 (Was able to do it with a different program since I had this issue, but I like dumldore and would like to figure out why it wont let me)
- I don't recall the exact message. my rndis ip was set up correctly and everything seemed in place. the message kept happening a little after pressing "flash firmware". it would start, and after a bit of loading it would say something about it timedout...I understand that it takes a little while for the program to fight throught the dji firmware so it can load, and im not sure if dumldore just didn't allow enough time.

TACK86, 2019-01-05:
I think I have you on slack now...…...

closed 2019-01-28 by jezzab