jezzab/DUMLdore #52: wont roll back from 01.04.500

scottyd13, 2019-06-19:
Hi, im pretty new to this modifying of parameters etc and stupidly made the mistake of updating my firmware to 01.04.500  .....Im trying to get it to roll back and and its pretty stuck.  Ive got no updates to roll back to in assistant either now.... any ideas ??

jezzab, 2019-06-19:
How about you start with what actual aircraft it is your trying to roll back.

scottyd13, 2019-06-19:
Sorry ..... mavic pro

crucialpro, 2019-06-24:
I'm having the same issue with my Mavic Pro Platinum. Won't downgrade to any of the older versions. Any advice please?

brett8883, 2019-06-28:
Guys you literally download whatever firmware you want to downgrade to using [https://github.com/cs2000/DankDroneDownloader](url) and then load that file into DUMLdore and then press flash. If you are downgrading to a firmware version that starts 01.03.xxxx you need to do this process TWICE.