jezzab/DUMLdore #53: "Dumldore has stopped working" error

esuohdica, 2019-06-25:
Hi Jezzab,

I'm stuck with Dumldore V3.20 unable to start on my home PC, i just get "Dumldore has stopped working" straight away. I've tried removing and reinstalling, but no success. I get the same message whether or not I have the Mavic connected.
What is Dumldore looking for as it starts? I'm sure this is either driver or port related, because everything works fine on a different laptop, it just frustrating that i can't fix this one!
I can FTP into the Mavic when its connected and assistant also connects no problem. Any ideas? Cheers
[image]
[image]

jezzab, 2019-06-25:
It sounds like a comport/serial port problem and the way the very old version of Windows you are running handles it. Have a look as some of the old issue requests for possible fixed

esuohdica, 2019-06-30:
Thanks Jezzab, i couldn't fix the issue on the original PC, but did get everything working on a second laptop (still windows7) cheers for the help, I think it is ADB driver issues, but I'm not looking any further into this as i've got a work around. Cheers

closed 2019-06-30 by jezzab