jezzab/DUMLdore #57: Serial port not found

pabloiarriola, 2019-10-22:
I am trying to flash the firmware of my mavic air remote control using the latest version of dumldore, but when I start it the error serial port not found. reconnect and restart shows up. The dji assistant does connect to it

jezzab, 2019-10-23:
The Mavic Air RC is upgraded via wifi not USB and is done by the Go App not DJI Assistant2

pabloiarriola, 2019-10-23:
Just that I do have another mavic air rc and I can update it using the dji assistant 2. It does show the firmware versions and everything. Im just looking for another way to update, because the one im having problems with does not show wich fw version its using

jezzab, 2019-10-23:
Sorry I was thinking Spark RC not Mavic Air. 
The Mavic Air RC is not supported.

pabloiarriola, 2019-10-23:
Oh ok ok yeah and is there another tool to flash the firmware to it?

wlbragg, 2022-03-24:
Same issue with Mini 2 on Com3, DUMLdore does not connect to serial port, DJI Assistant is off and DJI Drone on com3 appears in device manager. Any help here?
I'm still trying to correct gimble calibration error on 2 perfectly good working mini 2's.

jezzab, 2022-03-24:
Since the Mini 2 wasnt invented when this software was written, it is not supported.

wlbragg, 2022-03-24:
@jezzab  thank you for the reply.
 Are you no longer planning on updating this software?

Kopervlad, 2023-03-11:
Hi!
Same issue with Mini on Com10, DUMLdore does not connect to serial port, DJI Assistant is off and DJI Drone on com10 appears in device manager. Any help here?