jezzab/DUMLdore #61: ADB doesn't activate with P4P2 fw (v01.00.50.00)

StanLee4456, 2020-01-02:
Can Dumldore be updated where ADB will activate with the P4P2 fw?

knarfboy, 2020-01-06:
If P4P2.0 is upgraded to 01.00.50.00 firmware, it has rejected dumldore's adb function.

jezzab, 2020-01-06:
Im not surprised. I dont own a P4P v2 and Im not Nostradamus

knarfboy, 2020-01-06:
I afraid, all next DJI Firmware updates will reject DUMLdore

jezzab, 2020-01-06:
For DUMLdore to keep up with each firmware that comes out of each model it requires:
-Time
-Hardware

knarfboy, 2020-01-07:
Can you tell us which city / area you live in? Maybe there are people near you who can help you with hardware at least on loan.

StanLee4456, 2020-01-08:
I think NLD achieves ADB access ina different way than Dumldore. I guess there is no way to incorporate that method into a dumldore...

PLC-Fan, 2020-03-08:
Enable ADB doesn't work with firmware 01.00.00 either... :-(
What's the solution for P4PV2 owners? Can anyoine help?

primarytech, 2020-07-07:
> Enable ADB doesn't work with firmware 01.00.00 either... :-(
> What's the solution for P4PV2 owners? Can anyoine help?

Umm, I had no problem with Firmware V01.00.1500 ... ADB worked just fine.

knarfboy, 2020-07-07:
solution;<https://www.kopter-support.de/viewforum.php?f=101>

(v01.00.50.00) (#61)
Datum: 2020-07-07T20:37:56+0200
 
 
 
 

<https://github.com/jezzab/DUMLdore/issues/61#issuecomment-655048601> , or 
unsubscribe
"https://github.com/jezzab/DUMLdore/issues/61#issuecomment-655048601", 
"url": 
"https://github.com/jezzab/DUMLdore/issues/61#issuecomment-655048601", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]