jezzab/DUMLdore #63: 01.00.0620 fw mavic air can't reverse

cutis, 2020-03-12:
All ghosted options show in dumldore except get fw, get log fw last, flash
Gotta root 1st
But nothing auto pilots it
Try flash "root.bin" & others?

rtvater, 2020-03-12:
Works great thx cool

cutis, 2020-03-12:
Are you saying flash Root or NewRoot to access the dumldore auxiliary
functions ghosted?

	Without rooting all functions ghosted except get fw, flash fw, get
last fw log

	Intent is remove aircraft fw 01.00.0620 mavic air drone to fw
00.0X.0XXX which was zero generation fw as indicated by 1st two digit

	I am deficient understanding the evolution backpedaling. 

	I desire freeing my drone from capture interferes intruse.

	This will make my life flying much less humbug complicated.

	Please guide my freedom

	Thanks CurtisChun 

	-----------------------------------------From: "rtvater" 
To: "jezzab/DUMLdore"
Cc: "cutis", "Author"
Sent: Thursday March 12 2020 5:16:36AM
Subject: Re: [jezzab/DUMLdore] 01.00.0620 fw mavic air can't reverse
(#63)

	 Works great thx cool

 Sent from Yahoo Mail on Android 

 On Thu, Mar 12, 2020 at 1:46 AM, cutis wrote: 

 All ghosted options show in dumldore except get fw, get log fw last,
flash
 Gotta root 1st
 But nothing auto pilots it
 Try flash "root.bin" & others?

 —

[2]. 

Links:
------
[1]
https://github.com/jezzab/DUMLdore/issues/63#issuecomment-598244871
[2]

cutis, 2020-03-14:
> Works great thx cool Sent from Yahoo Mail on Android On Thu, Mar 12, 2020 at 1:46 AM, 

there's no documentation guide for flash strategy sequence "new.root.bin", "new.payload.bin" prior to flashing desired aircraft fw 00.01.0500 blocked by 01.00.0620 mavic air

jezzab, 2020-03-14:
As per the release notes of version DUMLdore 3.10 (https://github.com/jezzab/DUMLdore/releases/tag/v3.10)
Mavic Air root/ADB and Rollback were not supported from 01.00.0400 onwards and it wont be supported in the near future. 

The:
`payload.bin`
`payload_new.bin`
`root.bin`
`new_root.bin` 

Are not actually used and they are packed into the EXE file. I just forgot to remove them from the repo when I updated last.

vicmanhg, 2020-11-21:
What worked? did you manage to roll it back. i cant connect mi controler affter update. im very frustrated

vicmanhg, 2020-11-21:
> As per the release notes of version DUMLdore 3.10 (https://github.com/jezzab/DUMLdore/releases/tag/v3.10)
> Mavic Air root/ADB and Rollback were not supported from Mavic Air 01.00.0400 onwards and it wont be supported in the near future.
> 
> The:
> `payload.bin`
> `payload_new.bin`
> `root.bin`
> `new_root.bin`
> 
> Are not actually used anymore and they are packed into the EXE file. I just forgot to remove them from the repo when I updated last.

Have you heard how to fix the RC coneection problems? is there any mistery on why is there so liter information on why DJI is doing this. im tottaly locke out of mi AC...

marcuslindemannrohden, 2021-02-24:
Has anyone found a solution? I'm having an IMU Calibration error since I updated the last firmware version and not being able to fly it.