jezzab/DUMLdore #65: I have a mavic2 pro with firmware 01.00.510. What version do I need to roll back to, so I can install DJI assistant 1.1.2 and open parameters?

RENCG08, 2020-03-22:


xyls13, 2020-04-10:
That's impossible. I have the same machine. I've been looking for it for a long time. I don't have an old version that can downgrade the system version and adjust the parameters. Let's wait for the professionals to figure it out.

xyls13, 2020-04-11:
I have downloaded the historical firmware of mavic2pro, and also downloaded the 1.1.2 version of assistant 2. Do you want to try the self degraded version? If you want, I can provide you with all the tools and firmware you need. You can contact me at xueyulii@hotmail.com

my08rubicon, 2020-04-19:
I wrote you an email. I own a DJI Mavic Pro

xyls13, 2020-04-19:
> I wrote you an email. I own a DJI Mavic Pro

My machine model is: mavic2pro, our machine models are not the same, your mavic1pro can now use this author's software to brush the machine.

jezzab, 2020-04-19:
Mavic 2 01.00.0510 cannot be rolled back at this time.
This issue isn't related to the functionality of DUMLdore so im closing it

closed 2020-04-19 by jezzab

xyls13, 2020-04-19:
I've been demoted by myself. I use the latest version of assistant2 to demote successfully. Have I successfully downgraded to version 01.00.0300? Why not demote? Or the parameters cannot be modified in the degraded version?

 
Closed #65.