jezzab/DUMLdore #67: Mavic 2 Pro .510 to .400

cchin07, 2020-05-09:
was there anyway can help us to reach this hard problem?
Dji forced me to change new drone , it's only got .510
i need FCC...please help

cchin07, 2020-05-09:
my friend told me that, AC firmware you could still downgrade it yesterday, but today they cut it off , you no longer to downgrade from Smart Controller

mkeky, 2020-05-11:
please  help I also have the .510 fw and cannot downgrade ! :((

jezzab, 2020-05-11:
> my friend told me that, AC firmware you could still downgrade it yesterday, but today they cut it off , you no longer to downgrade from Smart Controller

.510 has always had anti rollback.

The method of delivery makes no difference. Its not the app or software that is used to flash the aircraft that blocks it, it is the software that is _on_ the drone. Using the smart controller, normal phone and Assistant2 etc will all net the same result, the delivery is the same, it cant just bypass it. 

Unfortunately I dont have a solution in the near future

closed 2020-05-11 by jezzab