jezzab/DUMLdore #69: No usb connection after flashing firmware mavic pro

dl2ffp, 2020-06-14:
DUMLdore has  during flashing firmware aborted. After that, no usb connection was possible.
The usb-port try to connect and in the same time disconnect. This is ongoing.
I suspect destroyed the bootloader in the mavic pro.
Any idea ?

jezzab, 2020-06-14:
DUMLdore does not flash the aircraft. It uploads the firmware via ftp and then tells the aircraft that it can begin an upgrade. The aircraft then upgrades itself by checking and reading the files that were uploaded via ftp. After the moment the upgrade begins the upgrade is autonomous. The aircraft can be unplugged from USB and will continue to upgrade itself, you just wont have any feedback on the process.

Without knowing what you flashed, what firmware you were on, what you were going to, where you got it etc. I cannot help you.

dl2ffp, 2020-06-14:
After booting the aircraft is not blinking yellow, is red. I wanted firmware downgrade from 04.400 of 04.300.

dl2ffp, 2020-06-14:
What can I do without usb? nothing or?

dl2ffp, 2020-06-14:
I probably switched off the aircraft during the flash process

jezzab, 2020-06-14:
Switching off the aircraft during the upgrade process could cause this and sounds more like it because going from 01.04.0400 to 01.04.0300 wont break a device. Even DJI Assistant can do this.

Im sorry but there is nothing I can suggest.

closed 2020-06-14 by jezzab