jezzab/DUMLdore #71: Unable to downgrade from v 1.04.0400 (Mavic Pro)

MavicPros, 2020-10-02:
Tried using DUMLdore to downgrade my Mavic Pro that is currently on version 1.04.0400.
When using the tool, the downgrade seems to work; it pauses for a while at 7%; then at 15%; and then it says it is complete.
But this does not actually downgrade the firmware; it just stays at 1.04.0400 - I've tried flashing multiple times.  Any ideas?

TheELFman, 2020-10-14:
I am having the same issue. Have tried over 20 times to flash to 1.04.300 and other firmware and the progress jumps to 7%, then 15%, then 20%, then all the way to 100% (Safe to power off) without actually flashing anything.

martgian, 2021-06-15:
Same issue here, did anyone find a solution?