jezzab/DUMLdore #78: Mavic 2 Zoom will not flash earlier release than 0790

andyalford, 2023-06-13:
I have tried to use your marvelous software to flash an earlier version than the latest 0790 with ARB 2.
I have tried a couple of BIN files from DDD but same experience with all, of course they have ARB of 0 or 1.
It replies that the Flash was successful, but it seems a bit quicker than I would expect.  
It takes a while to get to 7% then another while to 15% and then seems instant completion.
I suspect it simply is hitting the ARB and not reporting an error.  But I thought I would let you know.
Would the ARB be an unavoidable block?  Seems a bit naughty by DJI.  
Thanks for your work, and if there is anything that can be done, please let me know.

andyalford, 2023-06-13:
[image]
[image]
[image]

sakis08, 2024-01-22:
HELLO 
HAVE YOU FIND ANY SOLUTION YET ?