jezzab/DUMLdore #81: Mavic ADB enable question

K0zero, 2023-11-13:
I'm really interested in how 'adb enable' works. Could you explain the logic behind it? Thx!
I want to use a similar method to perform the same operation on my Mavic 3

kilrah, 2023-11-14:
The vulnerabilities used back then have been patched by DJI years ago, gonna have to find your own...

K0zero, 2023-12-29:
> The vulnerabilities used back then have been patched by DJI years ago, gonna have to find your own...

thx for your  comment, i just wanna know how this vulnerabilities works, i can't find the poc or any details about this exp.

jezzab, 2023-12-29:
> > The vulnerabilities used back then have been patched by DJI years ago, gonna have to find your own...
> 
> thx for your comment, i just wanna know how this vulnerabilities works, i can't find the poc or any details about this exp.

The binary isn't packed, you could easily deco pile the .Net binary (hence why I never packed it or posted the source). Decompile it and follow the bouncing ball, all the info is there for you to see

K0zero, 2024-01-04:
> > > The vulnerabilities used back then have been patched by DJI years ago, gonna have to find your own...
> > 
> > 
> > thx for your comment, i just wanna know how this vulnerabilities works, i can't find the poc or any details about this exp.
> 
> The binary isn't packed, you could easily decompile the .Net binary (hence why I never packed it or posted the source). Decompile it and follow the bouncing ball, all the info is there for you to see

yeah  I have tried using decompiling software like dnSpy, but there is a problem - I cannot parse the most important DUML library. I can't see any code related to this library, so it's difficult for me to debug the DUML commands transmitted via USB.
can you give some suggestion or advice ?  thx!

K0zero, 2024-01-04:
[image]

K0zero, 2024-01-04:
> The binary isn't packed, you could easily decompile the .Net binary (hence why I never packed it or posted the source). Decompile it and follow the bouncing ball, all the info is there for you to see

yeah!  I tried and I  did it.  thx for your suggestion!
so

closed 2024-01-04 by K0zero